nobleo/rviz_satellite #1: Does not compile in hydro

dani-carbonell, 2015-03-24:
I have Ubuntu 12.04 - gcc version 4.6.4 - ROS Hydro

I don't know how to solve this error:

cc1plus: error: unrecognized command line option ‘-std=c++11’

gareth-cross, 2015-03-24:
If you change it to `-std=c++0x` - does that resolve the issue?

dani-carbonell, 2015-03-24:
I've modified this line in the CMakeLists.txt:
```
add_definitions("-Wall -Wunused -std=c++0x")
```
and I get this errors:

```
####
#### Running command: "make -j8 -l8" in "/home/dardo/catkin_ws/build"
####
[ 16%] [ 50%] [ 50%] [ 66%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/tileloader.cpp.o
Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_tileloader.cxx.o
Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/aerialmap_display.cpp.o
Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_aerialmap_display.cxx.o
In file included from /usr/include/boost/signals/connection.hpp:13:0,
                 from /usr/include/boost/signals/signal_template.hpp:18,
                 from /usr/include/boost/signals/signal0.hpp:24,
                 from /usr/include/boost/signal.hpp:19,
                 from /usr/include/boost/signals.hpp:9,
                 from /opt/ros/hydro/include/tf/tf.h:45,
                 from /opt/ros/hydro/include/tf/transform_listener.h:38,
                 from /home/dardo/catkin_ws/src/rviz_satellite/src/aerialmap_display.cpp:27:
/usr/include/boost/signals/detail/signals_common.hpp:26:13: error: expected identifier before ‘protected’
/usr/include/boost/signals/detail/signals_common.hpp:26:13: error: expected unqualified-id before ‘protected’
In file included from /usr/include/boost/units/detail/utility.hpp:20:0,
                 from /usr/include/boost/exception/detail/type_info.hpp:19,
                 from /usr/include/boost/exception/detail/object_hex_dump.hpp:15,
                 from /usr/include/boost/exception/to_string_stub.hpp:16,
                 from /usr/include/boost/exception/info.hpp:16,
                 from /usr/include/boost/exception/detail/exception_ptr.hpp:20,
                 from /usr/include/boost/exception_ptr.hpp:9,
                 from /usr/include/boost/thread/future.hpp:14,
                 from /usr/include/boost/thread.hpp:24,
                 from /opt/ros/hydro/include/tf/transform_listener.h:43,
                 from /home/dardo/catkin_ws/src/rviz_satellite/src/aerialmap_display.cpp:27:
/usr/include/c++/4.6/cxxabi.h:47:37: error: expected ‘}’ before end of line
/usr/include/c++/4.6/cxxabi.h:47:37: error: expected declaration before end of line
make[2]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/src/aerialmap_display.cpp.o] Error 1
make[1]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed
```

Which ROS version are you using?

gareth-cross, 2015-03-24:
I am using Indigo (1.11.10) on Ubuntu 14.04, with gcc 4.8.2. I do not have a Hydro machine handy at the moment. Historically I have had difficulty using the c++11 switch on Ubuntu 12, since it requires upgrading both gcc and boost.

dani-carbonell, 2015-03-24:
Thank you gareth for the fast answer.
I've just installed the package on Ubuntu 14.04 - Indigo and it compiles perfect. If I am able to install it on hydro I will let you know. By the moment I am interested in the functionality so I am testing it on Indigo.

closed 2015-04-20 by gareth-cross