nobleo/rviz_satellite #2: QImageLoader Throwing Error

mikehosmar, 2015-04-07:
I have been trying to get this package to work on my machine but I'm getting an error when the tiles are being downloaded.  I have tried poking around to see what's wrong but I'm not familiar with Qt so I've hit a bit of a wall.
I am running Ubuntu 14.04 with a full desktop install of ROS Indigo.

The issue is here: https://github.com/gareth-cross/rviz_satellite/blob/master/src/tileloader.cpp#L139

The QImagereader object throws a `Reading ras files from sequential devices not supported` error from within the if statement. (btw the if statement was still returning true) This stack overflow topic seems to be the same issue but I wasn't able to emulate the fix.  He only needed to install the libjpeg.so.62 library to fix his problem.  I checked and only had version 8 installed on my machine so I installed 6.2 on top.  I have no idea if it worked however.
http://stackoverflow.com/questions/11926840/qt-jpg-image-display-with-qlabel

I tried this as well which would prevent the error but the program still wouldn't work.  It would flip frantically between "tiles fully loaded" and "loading tiles"
https://code.google.com/p/qimagegrabber/issues/detail?id=2

Let me know if you need any more info or if you have any ideas.

Mike Hosmar

gareth-cross, 2015-04-08:
Hi Mike, apologies for the delayed reply. Are the URLs (to the images) themselves valid? They should appear in the console when you run rviz. Try pasting it into a browser and see if it corresponds to a real tile.

I am also running Ubuntu 14.04 w/ ROS Indigo but I have not experienced this issue yet.

mikehosmar, 2015-04-08:
Hey Gareth,
I did try that. At the default zoom level it wasn't valid but if I change the zoom level I still have the same issue.

I'm thinking that the library dependencies are my issue.  Do you have qt creator installed?  maybe libjpeg was installed during your development.

I wanted to check my installed packages.  Here is the output of `dpkg --get-selections | grep -v deinstall`.
I think libjpeg is around line 1250.  It may not be much help but take a look.

http://pastebin.com/gzSL8y15

Ps while this is running im not getting images but I must be downloading them b/c im running a consistent 2.5 MB/s some times 8+ MB/s.

mikehosmar, 2015-04-08:
Actually the effects of the settings in rviz are a bit more complex

1. When first added it does nothing because there is no topic
2. After a topic is selected the node starts downloading images in the area of http://otile1.mqcdn.com/tiles/1.0.0/sat/16/34386/22251.jpg
These are not valid.  The indicator here is that the resolution is set to 0 and isnt changing.  The Reading Ras error starts and there seems to be 2 per tile loaded.
3. If I change the zoom level to 10 the images are now valid are are in the area of http://otile1.mqcdn.com/tiles/1.0.0/sat/10/538/345.jpg The resolution is now a reasonable number and my internet usage spikes up to 8.5MB/s. Still no image and the errors continue.
4. If I reduce the number of blocks it flicks back and forth between finished loading and loading.

This makes me think the images are being downloaded and some how being marked as not valid.  This could be the jpeg library or the image reader.

gareth-cross, 2015-04-09:
Here is the output on my machine: http://pastebin.com/tS3HZNtq

It looks like I only have `libjpeg8` installed. What Qt version are you on? Mine is 4.8.6.

mikehosmar, 2015-04-10:
I also have 4.8.6. This might not be the problem I thought it was.  Do you know any other map sources that might have the tiles in .png? Testing it with that will show if the jpeg libraries are to blame.

mikehosmar, 2015-04-15:
I still havn't gottent it to work.  Do you mind trying to do the exact same thing I am?

I am using gazebo to simulate clearpaths jackal.  They have a gps simulator built into their example and I havn't changed their default start location.  I am just running the example gazebo launch file. launching the jackals rviz config and adding your plugin.  Once I do that I change the zoom to 10 then changing the topic to /navsat/fix.  And I get nothing.

The process may go something like this:

```
sudo apt-get install ros-indigo-jackal-*

roslaunch jackal_gazebo jackal_world.launch

roslaunch jackal_viz view_robot.launch
```

gareth-cross, 2015-04-20:
Hi @mikehosmar, sorry for the slow reply - I have been very busy lately. Not sure if you have made further progress since your last post. I tried the steps you provided. 

At the default zoom level (16) and Object URI there is no imagery for the starting point of the simulation (lat = 49.90001127, lon = 8.89998717). At zoom level 10 there is imagery. I have corrected the issue where the imagery would continually re-load, this was a simple mistake. Try pulling the fix/reload-issue branch. With this code and zoom level 10 the simulation loads imagery.

mikehosmar, 2015-04-21:
Hey, 
Your branch seems good. it works on my end.

I am still seeing the `Reading ras files from sequential devices not supported` error and the display is saying that there isnt a transform from world to odom.  But it does work.

Thankyou,

Mike Hosmar

mikehosmar, 2015-04-21:
Hwen I move around the map the map seems to shift 90 degrees from the direction that I'm moving.  Any Ideas of why that is?

schra, 2019-03-12:
@mikehosmar I'm closing this issue for now. If this issue still applies to the current version, I'll reopen this issue again.

closed 2019-03-12 by schra