nobleo/rviz_satellite #5: Extend NavSatFix msg

jpardeiro, 2015-07-29:
Hi Gareth,

For my application I need to be able to reconfigure the zoom value by a topic instead of using the field in RViz. The way to solve that could be a modification of the NavSatFix msg for adding this parameter, or create a new one which will have just all the fields needed for your plugin. I want to ask you if there is any reason to not include this feature, or just nobody has need this before.
 
Great work!

Regards,

Jose

gareth-cross, 2015-07-30:
It has not been necessary so far, although I can imagine why it might be useful. NavSatFix is part of the ROS framework, and cannot be modified so easily. Instead I can recommend two options:

1) Create a second topic to which the plugin can subscribe, and publish zoom levels on that topic.
2) Create a ROS service in the plugin which can be called externally by other nodes (or the user at the command line) and change the zoom level using that.

chataign, 2016-01-07:
I think using a service would be the more natural way to solve this, though using a topic would potentially be faster as it is non-blocking. I can't really think of a use-case for this feature, but if I'm happy to implement it if needed.

schra, 2019-03-12:
@jpardeiro We don't have this use case and therefore will not implement this ourselves. However you are welcomed to provide a PR for your feature request.

closed 2019-03-12 by schra