nobleo/rviz_satellite #15: Render Level for Draw Behind

tdenewiler, 2015-12-17:
In `rviz` when I have the `AerialMapDisplay` running with `Draw Behind` set to true and then I add a `Map` display type with `Draw Behind` also set to true the two layers swap which is on top. It seems that this is based on the height of the data in the `Map` topic. I was able to make the `AerialMapDisplay` always be behind the `Map` display type by changing `obj->setRenderQueueGroup(Ogre::RENDER_QUEUE_4);` to `obj->setRenderQueueGroup(Ogre::RENDER_QUEUE_3);`.

I believe the reason the display types fight each other is that the `Map` display type also uses the value of `4` (seen [here](https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/default_plugin/map_display.cpp#L296)).

beetleskin, 2019-03-12:
Setting this to `Ogre::RENDER_QUEUE_3` is your personal preference, others might prefer something else. @schra this wont fix

schra, 2019-03-12:
@beetleskin This was already merged in #28

closed 2019-03-12 by schra