nobleo/rviz_satellite #21: Segfault when loading satellite to rviz

davidrotor, 2016-06-15:
when I try to load the display it segfaults on ubuntu 16.04 with kinetic

chataign, 2016-06-15:
Hi David,

Are you saying that `rviz_satellite` is crashing instantly, even before you can configure it in `rviz`?

Could you run `rviz` with GDB to see if we can catch something in the backtrace?

http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Thanks

davidrotor, 2016-06-15:
ok I ran it with gdb and got this result

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007fff771025a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4

chataign, 2016-06-15:
Ok and what does it say if you type "bt" (backtrace) in gdb?

davidrotor, 2016-06-17:
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007fff79c865a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) bt
#0  0x00007fff79c865a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7de74ea in call_init (l=<optimized out>, argc=argc@entry=1,
    argv=argv@entry=0x7fffffffdad8, env=env@entry=0x7fffffffdae8)
    at dl-init.c:72
#2  0x00007ffff7de75fb in call_init (env=0x7fffffffdae8,
argv=0x7fffffffdad8,
    argc=1, l=<optimized out>) at dl-init.c:30
#3  _dl_init (main_map=main_map@entry=0x1876be0, argc=1,
argv=0x7fffffffdad8,
    env=0x7fffffffdae8) at dl-init.c:120
#4  0x00007ffff7dec712 in dl_open_worker (a=a@entry=0x7fffffffbb70)
    at dl-open.c:575
#5  0x00007ffff7de7394 in _dl_catch_error (
    objname=objname@entry=0x7fffffffbb60,
    errstring=errstring@entry=0x7fffffffbb68,
    mallocedp=mallocedp@entry=0x7fffffffbb5f,
    operate=operate@entry=0x7ffff7dec300 <dl_open_worker>,
    args=args@entry=0x7fffffffbb70) at dl-error.c:187
#6  0x00007ffff7debbd9 in _dl_open (
    file=0x184c6b0 "/home/david/Paquetes/devel/lib//librviz_satellite.so",
    mode=-2147483391,
    caller_dlopen=0x7ffff111d862
<Poco::SharedLibraryImpl::loadImpl(std::__cxx11::basic_string<char,
std::char_traits<char>, std::allocator<char> > const&)+82>, nsid=-2,
argc=<optimized out>, argv=<optimized out>, env=0x7fffffffdae8)
    at dl-open.c:660

davidrotor, 2016-06-18:
is it fixable? im using ros kinetic

chataign, 2016-06-21:
Hi, I was hoping that gdb would give us more info but there isn't much in the debug output you sent.

I don't have a setup running Ubuntu 16.04 so I'm not able to test I'm afraid... Is there any chance you have a 14.04 setup? So you can confirm that the plugin is running OK? I have been using it with 14.04 + ROS jade and it's been working great

davidrotor, 2016-06-21:
I hace finales this week so Ill try to vmware it friday night and try it out

nxdefiant, 2017-12-22:
Confirmed for lunar and probably all later ros distros. CMakeLists.txt only checks for kinetic in line 8.

commit 0872233e (2018-12-04): Detect Qt 5 automatically, adding support for ROS Lunar, Melodic, et al

Based on https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_rviz_plugins/CMakeLists.txt

Fixes #21, #40, #43, #45
closed 2018-12-04 by schra