nobleo/rviz_satellite #24: Only map levels up to 11 are properly retrieved from MapQuest for some locations

skohlbr, 2016-07-12:
When trying out the plugin for a location I'm interested in, I noticed that it only properly downloads map data from level 1 to 11. All others show grey image tiles with a crossed out camera. When viewing the same location on map quest, it is obvious higher resolution data than level 11 is available: http://mapq.st/29F0sQV

It seems that MapQuest did some API changes (the old URL says something that in July 2016 they switch to a new site). So maybe what I observe is related to that. Any hints are appreciated, at map level 11 is basically unusable.

Here's a bagfile with NavSatFix data that can be used to reproduce the issue:
[2016-07-11-20-27-13.bag.zip](https://github.com/gareth-cross/rviz_satellite/files/358729/2016-07-11-20-27-13.bag.zip)

skohlbr, 2016-07-12:
Fun stuff, just tested in simulation and now get this:
[image]

It seems this is a bigger problem with MapQuest that could require some major changes.

skohlbr, 2016-07-12:
Here's the blog post: http://devblog.mapquest.com/2016/06/15/modernization-of-mapquest-results-in-changes-to-open-tile-access/

tl:dr: One needs to register now to still be able to access tiles. 15000 monthly transactions are free.

skohlbr, 2016-07-12:
These tile URLs worked as a quick fix for me (not satellite, but better than nothing):
http://tile.stamen.com/watercolor/{z}/{x}/{y}.jpg
http://tile.stamen.com/toner/{z}/{x}/{y}.png
http://a.tile.thunderforest.com/landscape/{z}/{x}/{y}.png

gareth-cross, 2016-07-12:
Did a quick investigation into this. Seems like even w/ the MapQuest key, the old tile URL does not work anymore. They _do_ still provide tile access through the Leaflet/JavaScript API, documented [here](https://developer.mapquest.com/documentation/leaflet-plugins/migration-guide). 

However, upon further inspection, the access point for their JavaScript API (listed as MapQuest JavaScript SDK) simply converts the key into a MapBox access token and returns a wrapper script that queries MapBox. Apparently MapBox now backs all the tile requests for MapQuest. 

No point in going through MapQuest anymore. With a MapBox account (mine is currently free trial) I am able to access the satellite imagery w/ the URL: 

`http://a.tiles.mapbox.com/v4/mapbox.satellite/{z}/{x}/{y}.jpg?access_token=<TOKEN>` 

where `<TOKEN>` is my public access token, accessible from the `API Access Tokens` section of the account page. 

Let me know if that works for you and we can update the documentation.

skohlbr, 2016-07-13:
Thanks for the update! I can confirm things work for me using the URL with the access token at the end.

closed 2016-07-15 by gareth-cross