nobleo/rviz_satellite #38: Use this plugin with a transform or Odometry

reinzor, 2018-02-28:
Hi, 

We are fusing various sources to obtain our final position estimate in UTM frame. Can we use this plugin without a NavSat message? If not, could you provide me some pointers for implementing this? 

Thanks

beetleskin, 2019-03-07:
@reinzor I guess this is outdated?

To answer the question: of course, you can fork this plugin and use your custom-message instead of the NavSat one.

reinzor, 2019-03-12:
Thanks for your reply, this is indeed outdated for me so I will close the issue.

closed 2019-03-12 by reinzor

Timple, 2019-11-29:
There is a workaround if you use the [navsat_transform_node](http://docs.ros.org/melodic/api/robot_localization/html/navsat_transform_node.html). You can set the `~publish_filtered_gps` parameter which will output a `/gps/filtered` topic which will contain your fused position in gps notation which you can then use for this plugin.