nobleo/rviz_satellite #46: How to replace openstreetmap with Google Map

nofear123, 2019-01-15:
Compared with Google,OpenStreetMap and MapBox are updated slowly.So I found the URI of Google tiles Maps on the Internet. Then I directly replaced the object uri with the uri of Google Maps on AerialMapDisplay.
In the end, although Google Maps can be displayed on the rviz, the scale of the map is incorrect, **so some people know how to modify the scale**?

Thank you.

beetleskin, 2019-03-07:
what link-template and other settings are you using?

nofear123, 2019-03-09:
This is my link to use the Google map, you can try it.This link has a lot of parameters to adjust.
http://mt2.google.cn/vt/lyrs=y&scale=1&hl=zh-CN&gl=cn&x={x}&y={y}&z={z}

beetleskin, 2019-03-12:
it works for me with http://mt0.google.com/vt/lyrs=s&hl=en&x={x}&y={y}&z={z}&s=Ga

Maybe play a bit with the other params?
However, using those links without api is deprecated and might have some legal issues. Just fyi.

nofear123, 2019-03-12:
Yes, your link is useful. I compared your link to my link  and modified some params. I found the parameter in my link that caused my problem is "&gl=cn". Thank you for your answer.

nofear123, 2019-03-12:
> it works for me with [http://mt0.google.com/vt/lyrs=s&hl=en&x={x}&y={y}&z={z}&s=Ga](http://mt0.google.com/vt/lyrs=s&hl=en&x=%7Bx%7D&y=%7By%7D&z=%7Bz%7D&s=Ga)
> 
> Maybe play a bit with the other params?
> However, using those links without api is deprecated and might have some legal issues. Just fyi.

@beetleskin Thanks for your answer.

closed 2019-03-12 by nofear123

atasoglou, 2021-03-15:
Is the google link still working to date? We are getting "Bad request" errors.