nobleo/rviz_satellite #48: Tile Loader not showing the tiles in the proper place

david-alejo, 2019-03-13:
Dear all,

First of all, thank you for the development of this plugin that I've found very handy.

I upgraded to the latest version yesterday to make it work in 18.04. I have found that now the satellite images are no longer in the proper placed. Instead, they appear with a small shift to the South and East.

I attach an example with the representation of coordinates 41.38738993 N  2.183433017 E in google maps and with your plugin in rviz to help to understand the differences.

Has any other noticed it?
[image]
[image]

beetleskin, 2019-03-19:
Is your robot_frame equal to the frame_id in the NavSat message?

Could you provide a minimum working example to reproduce, via a PR?

david-alejo, 2019-03-21:
Thank you for your kind response. I'm checking the frame_id and if I can't correct it, I will provide a MWE.

Regarding to the tile served, I checked Google Maps and OpenStreetMap.

david-alejo, 2019-03-21:
I created a PR related to this issue as a minimum working example. In it, you can see a marker and the axes in the reference frame map. They do not correspond with the location of /gps/fix when using google maps.

beetleskin, 2019-03-22:
Well, the demo is currently not working properly, the display produces an error. In your PR, please add

`<node pkg="tf2_ros" type="static_transform_publisher" name="static_transform_base_link_applanix" args="0 0 0 0 0 0 map base_link" />
` and set the robot_frame to base_link. And try again.

Can you confirm that this works? If so, this is a duplicate of #49.

david-alejo, 2019-03-22:
Interesting. By doing so the tiles are loaded OK.

However, it is not clear why this is necessary. I mean: the plugin was configured to have /map as robot frame, why is it necessary to use the base_link frame?

beetleskin, 2019-03-22:
Its not necessary (I think) - but some frames need to be defined. That's true for most Display plugins, and for RViz itself.

david-alejo, 2019-03-22:
Ok it makes sense, I'm closing the issue as it is somehow duplicated with yours.

Thank you for your kind responses. Best regards,

closed 2019-03-22 by david-alejo

beetleskin, 2019-04-23:
@schra maybe this isn a pure duplicate - I think the plugin should complain if frames are missing, rather than rendering bullshit. What do you think?

schra, 2019-04-29:
> @schra maybe this isn a pure duplicate - I think the plugin should complain if frames are missing, rather than rendering bullshit. What do you think?

Good point. I created an issue for this.