nobleo/rviz_satellite #51: Don't show map if transform is missing

schra, 2019-04-29:
If AerialMapDisplay/Status/Transform shows the error `Fixed Frame [map] does not exist` then the map shouldn't be drawn at all. This leads to confusing results, see #48

beetleskin, 2019-04-30:
Could be done by using [MessageFilterDisplay](http://docs.ros.org/jade/api/rviz/html/c++/classrviz_1_1MessageFilterDisplay.html), like [most display plugins](https://github.com/ros-visualization/rviz/tree/melodic-devel/src/rviz/default_plugin) do.

m-naumann, 2019-05-18:
I encountered the same while working on #55

The easiest would be to set `scene_node->setVisible(false);` but the problem I see is that there are cached tiles that are made invisible, and when setting `scene_node->setVisible(true);` after successful transformations, they are all set to true, so also old/irrelevant tiles are shown.

schra, 2020-04-28:
From #72:

> When you set an invalid ff, for example "asdfasdfsf", then the map will jump. (It will not jitter, but jump). The status shows that there is an error, but the map is still displayed. I think we should hide the map if the ff is invalid?

Timple, 2026-10-25:
Can you try https://github.com/nobleo/rviz_satellite/pull/148 ?

Sorry for the delay 😄