nobleo/rviz_satellite #54: Unable to select fixed frame

RonaldEnsing, 2019-05-15:
The AerialMapDisplay plugin sets the fixed frame in rviz to `map` [1]. As a user I would expect that I can select the fixed frame in rviz. The fact that this plugin resets the fixed frame back to `map` every time I try to change the fixed frame is confusing.

The AerialMapDisplay ensures that the visualization will be drawn with the right position and orientation after changing the fixed frame to `map`. However, I think it would be really useful to be able to visualize the AerialMapDisplay with the right position and orientation, for any other fixed frame as well. This would mean that the additional transformations have to be applied between `map` and `fixed_frame_` to get the AerialMapDisplay on the right place.

[1] https://github.com/gareth-cross/rviz_satellite/blob/master/src/aerialmap_display.cpp#L515

RonaldEnsing, 2019-05-17:
See this commit [2] to allow any selectable fixed frame in rviz. This does require your `tf` tree to have a `utm` frame, also to get the orientation of the tile with respect to your robot.

[2] https://github.com/tud-cor/rviz_satellite/tree/799e65d6800677f67737b6a776de3a2454cf1e11

beetleskin, 2019-05-20:
> The fact that this plugin resets the fixed frame back to `map` every time I try to change the fixed frame is confusing.

Could you elaborate this a bit? How is the plugin resetting the fixed frame?

> The AerialMapDisplay ensures that the visualization will be drawn with the right position and orientation after changing the fixed frame to `map`. However, I think it would be really useful to be [...]

You can already select any frame in RViz, the map will transform accordingly assumed that `map` is your UTM frame, i.e. is aligned as you define your global coordinate system (e.g. ENU).

RonaldEnsing, 2019-05-20:
> Could you elaborate this a bit? How is the plugin resetting the fixed frame?

The fixed frame is set to `map` here https://github.com/gareth-cross/rviz_satellite/blob/master/src/aerialmap_display.cpp#L515

> You can already select any frame in RViz, the map will transform accordingly assumed that `map` is your UTM frame, i.e. is aligned as you define your global coordinate system (e.g. ENU).

Removing that line would not be enough.

schra, 2019-05-20:
I can confirm this bug and will fix it as soon as possible.

closed 2020-04-17 by schra

Kilswixh, 2025-04-05:
Please, what about Jazzy? I am facing the same problem.
I have no options; I cannot even select map

PR #59 (2020-04-17): Fix setting the fixed frame
Instead of setting the fixed frame, do the necessary rotation and transformation manually.

Fixes #54

/cc @beetleskin