nobleo/rviz_satellite #57: Support for Qt 5 [FeatureRequest]

yossioo, 2019-06-05:
Compiling with catkin build on Kubuntu 18.04+Melodic:
Found unsuitable Qt version "5.9.5" from /usr/bin/qmake, this code requires Qt 4.x

Any plans to adapt to Qt 5?

schra, 2019-06-05:
We are already supporting Qt 5. See the `CMakeLists.txt`.

I also just checked this and succeeded compiling the plugin with the following setup:
* Ubuntu 18.04.2
* ROS melodic
* Qt 5.9.5

Please check your Qt installation and check that `rviz_QT_VERSION` inside of `CMakeLists.txt` is really equal to `5.9.5`.

yossioo, 2019-06-05:
Well, for some reason I had some really old version of this package:`<version>0.0.2</version>`
Cloning fresh resolved the problem.

closed 2019-06-05 by yossioo