nobleo/rviz_satellite #60: ow to calculate the x,y-tile-coordinates from lat,long if given images in EPSG25832-projection method?

autonomobil, 2019-09-26:
Hello there,

I have a question regarding the creation of the WMTS server requests. 

From [this](https://www.bezreg-koeln.nrw.de/brk_internet/geobasis/luftbilderzeugnisse/digitale_orthophotos/index.html) german governmental-website I can get the address for their WMTS server: https://www.wmts.nrw.de/geobasis/wmts_nw_dop. I can successfully integrate it into josm as an image layer (wmts:https://www.wmts.nrw.de/geobasis/wmts_nw_dop) and it gets displayed correctly.

I can watch the server request made by josm in the terminal, which is something like:`2019-09-26 18:54:31.651 INFO: GET https://www.wmts.nrw.de/geobasis/wmts_nw_dop/tiles/nw_dop/EPSG_25832_16/16/17680/35107 -> HTTP_1 200 (13.3 kB)`
You can view this tile image: https://www.wmts.nrw.de/geobasis/wmts_nw_dop/tiles/nw_dop/EPSG_25832_16/16/17680/35107

In the rviz-satellite-plugin I entered the string for the request: https://www.wmts.nrw.de/geobasis/wmts_nw_dop/tiles/nw_dop/EPSG_25832_16/{z}/{y}/{x}
If I let the rviz_satellite figure out the y and x, I'm getting trash in rviz:
[image]

I think this is because the images/tiles are in EPSG25832-projection method, but the rviz-satellite thinks in Mercator-projection. And I need the images to be in EPSG25832, because of other data which gets displayed.

If you look at the WMTSCapabilities.xml of the map-server you can see there are images available up to zoom level 16 and other info: https://www.wmts.nrw.de/geobasis/wmts_nw_dop/tiles/nw_dop/EPSG_25832_16/1.0.0/WMTSCapabilities.xml

I had a look at [the methods, which the rviz-satellite seems to use to calculate the x and y](https://wiki.openstreetmap.org/wiki/Slippy_map_tilenames#Lon..2Flat._to_tile_numbers), which is something like this:
```
// in Coordinates.h:
int const n = 1 << zoom;
double x = n * ((coord.lon + 180) / 360.0);
double y = n * (1 - (std::log(std::tan(lat_rad) + 1 / std::cos(lat_rad)) / M_PI)) / 2;
```

But I think I need to do something like this, because of the EPSG25832/UTM projection:
https://en.wikipedia.org/wiki/Transverse_Mercator_projection

I already had a look at https://josm.openstreetmap.de/doc/org/openstreetmap/gui/jmapviewer/Tile.htm to see how openstreetmap does it, but I didn't find anything.

**So all in all: How to calculate the x,y-tile-coordinates from lat,long if given images in EPSG25832-projection method (which josm/osm does succesfully)?**

schra, 2019-09-27:
Hey Moritz,

could it be that the tiles are just twisted wrong? Would the tiles be correct if you rotate them 90° to the right and then mirror it along the x-axis (so that the old top of the tile is the new bottom)? I have a line in the code that mirrors some axis: https://github.com/gareth-cross/rviz_satellite/blob/40bc9c5caf4fee1079333d37792e029da082700c/src/aerialmap_display.cpp#L416-L419
You could try to adjust that line if you didn't already.

schra, 2019-10-07:
I'm closing this issue for now.

closed 2019-10-07 by schra