nobleo/rviz_satellite #61: Rotate the map image

weisongwen, 2019-10-24:
Hi @gareth-cross  Thanks for sharing this meaningful package. 

How can I rotate the map image in the Rviz ? for example rotate 45 degrees with z axis.

Best,
Welson,

schra, 2019-10-25:
Hey Welson, this plugin doesn't provide such a functionality. You have to adjust the source code for that. Could you elaborate on your use case?

schra, 2019-11-01:
Closed due to inactivity

closed 2019-11-01 by schra

weisongwen, 2019-11-04:
Hi @schra 
I have just fixed this problem by set rotation to the scene_node_. The code is added as follows:
`
Ogre::Matrix3  xyz_R_ned;
  Ogre::Radian radx(Ogre::Real(0*  3.1415926 /180));
  Ogre::Radian rady(Ogre::Real(0* 3.1415926 /180));
  Ogre::Radian radz(Ogre::Real(rotateAngleinDegree * 3.1415926 /180));
  xyz_R_ned.FromEulerAnglesXYZ(radx,rady,radz);
  position = xyz_R_ned * position;
  scene_node_->setPosition(position);

  int const convention = frame_convention_property_->getOptionInt();
  if (convention == FRAME_CONVENTION_XYZ_ENU)
  {
    scene_node_->setOrientation(Ogre::Quaternion::IDENTITY);
    Ogre::Matrix3  xyz_R_ned;
    Ogre::Radian radx(Ogre::Real(0*  3.1415926 /180));
    Ogre::Radian rady(Ogre::Real(0* 3.1415926 /180));
    Ogre::Radian radz(Ogre::Real(rotateAngleinDegree * 3.1415926 /180));
    xyz_R_ned.FromEulerAnglesXYZ(radx,rady,radz);                             
    scene_node_->setOrientation(xyz_R_ned);
  }
`
The variable `rotateAngleinDegree` is the angle you want to rotate the map with respect to the z axis. 
Thanks again for sharing this package.

Best,
Welson,

Hisham178, 2022-10-07:
Hi @weisongwen 

Thanks for your code. I managed to rotate the map from the center. I refactor a bit of your code.

    

>     Ogre::Matrix3  xyz_R_ned;
>     Ogre::Radian radx(Ogre::Real(0*  3.1415926 /180));
>     Ogre::Radian rady(Ogre::Real(0* 3.1415926 /180));
>     Ogre::Radian radz(Ogre::Real(90 * 3.1415926 /180));  //rotate 90 degrees  map to the right
>     xyz_R_ned.FromEulerAnglesXYZ(radx,rady,radz);   
>                           
>     auto const t_centertile_fixed = t_fixed_map + xyz_R_ned * t_centertile_map_;
> 
>     scene_node_->setPosition(t_centertile_fixed);
>     scene_node_->setOrientation(xyz_R_ned);