nobleo/rviz_satellite #67: Bug in the demo ?

tomlogan501, 2020-02-18:
Hi,
I try to launch the demo.launch and this is what happen : 

> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: melodic
>  * /rosversion: 1.14.3
> NODES
>   /
>     fake_gps_fix (rostopic/rostopic)
>     rviz (rviz/rviz)
>     static_tf_fake (tf2_ros/static_transform_publisher)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[fake_gps_fix-1]: started with pid [27967]
> process[rviz-2]: started with pid [27968]
> process[static_tf_fake-3]: started with pid [27969]
> /opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py:1826: YAMLLoadWarning: calling > yavml.load_all() without Loader=... is deprecated, as the default Loader is unsafe. Please read > https://msg.pyyaml.org/load for full details.
>   for d in data:
> terminate called after throwing an instance of 'std::invalid_argument'
>   what():  The number of blocks has to be positive

I'm using only source of ROS because I'm on 19.10 Ubuntu.

How can I fix this ?

iampy1, 2020-02-20:
I have the same issue. Please provide solution.

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    fake_gps_fix (rostopic/rostopic)
    rviz (rviz/rviz)
    static_tf_fake (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [7116]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d0c54dfe-53ce-11ea-b968-0800277151de
process[rosout-1]: started with pid [7131]
started core service [/rosout]
process[fake_gps_fix-2]: started with pid [7134]
process[rviz-3]: started with pid [7135]
process[static_tf_fake-4]: started with pid [7148]
terminate called after throwing an instance of 'std::invalid_argument'
  what():  The number of blocks has to be positive

iampy1, 2020-02-20:
Hello!

I found the solution although could not quite figure out why the error appears the way it does.

So, in file 'demo.rviz', under 'Visualization Manager -> Blocks' change the value from 3 to 8 (this the maximum value for the block field). Now run the 'demo.launch' again. It works!

PS: Once the rviz opens up, on the left side you can change the value of 'Blocks' to 3 again if you want.

Cheers! :)

schra, 2020-04-02:
Thanks for reporting this issue, `demo.launch` works again :)

commit dd2d6862 (2020-04-02): Fix race when reading the blocks property

This diff fixes the demo.launch that didn't work because of this bug.
Fixes #67
closed 2020-04-02 by schra