nobleo/rviz_satellite #69: Create branch for ROS2 port

blacksoul000, 2020-04-07:
I did the initial port to Dashing.
https://github.com/blacksoul000/rviz_satellite
Some problems with console output but it compiles and works.
Thanks

schra, 2020-04-17:
Thanks for the initial porting! We are thinking about providing a ROS2 branch, but I'm not sure to what extend I can provide maintenance for it. I'm currently stuck with ROS1, so the main development will definitely continue on the ROS1 version for now. I'm also not really sure if it's feasible to provide both ROS1 and ROS2 versions in parallel? Anyway, I will post an update here if there is an update about ROS2 and thanks again for your effort.

beetleskin, 2020-05-11:
@blacksoul000 if you could rebase towards #79 (soon to be integrated), we could create a dashing/foxy branch to use.
Related to #80.

blacksoul000, 2020-06-09:
> @blacksoul000 if you could rebase towards #79 (soon to be integrated), we could create a dashing/foxy branch to use.
> Related to #80.

rebased and works.

beetleskin, 2020-06-13:
@schra would you help out, and create an eloquent/foxy branch?

schra, 2020-08-02:
@blacksoul000 I added a note about your repository in the README in 83f093d96f4a4baf43b8c79716ae499804dc3738. Thanks again for your effort :)

@beetleskin I won't integrate any ROS 2 code in this repository yet

closed 2020-08-02 by schra