nobleo/rviz_satellite #78: Can we use Github actions for testing and linting?

beetleskin, 2020-04-19:
.. or should we rather stick with the ROS buildfarm hooks?

However, we should incorporate basic CI checks, linting and unit-testing.

schra, 2020-04-19:
Yeah this is something that we definitely should do. I also think integration into the ROS buildfarm is something we should do anyway.

It seems that there is no GitHub-integration of the ROS buildfarm? If you look at a ROS repo, for example https://github.com/ros-planning/moveit, they always use Travis.

I'm not familiar with both GitHub actions or Travis, but I'll have a look at this.

schra, 2020-05-05:
Idea from #79:

> what about clang-format and camelCase vs. under_score? Can one set that in the config? Or is that only something clang-tidy can do? Should we add a clang-tidy config? Is there one for ROS? I only know the one from moveit (https://github.com/ros-planning/moveit/blob/master/.clang-tidy).

Timple, 2021-01-19:
Enabled ros_industrial CI with catkin_lint

closed 2021-01-19 by Timple