nobleo/rviz_satellite #80: Melodic release

Timple, 2020-04-20:
I've noticed a lot of refactoring being done. Glad to see this code is maintained!

Would you also be interested in creating a release for melodic so we don't need to build from source?

Thanks for considering!

schra, 2020-04-22:
Not totally sure what you are referring to with a Melodic release.

But regarding not building from source: Integration into the ROS build farm is something that we might do in the future. What I can definitely do however, is to attach a .deb file to next releases.

Timple, 2020-04-23:
Integration into the ROS build farm is exactly what I meant. It is like a 2 hour job the [first time](http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease) you do it. After that pushing updates is easy.

No need to attach .deb files, that would be a-typical from all other ROS packages and (thus) we wouldn't use that.

Thanks for considering!

Timple, 2020-05-22:
Today a [call for testing](https://discourse.ros.org/t/ros-noetic-call-for-testing/14203) was done on discourse for Noetic, so when this issue is picked up, please also release for Noetic :slightly_smiling_face:

Timple, 2020-12-22:
Would you mind if I release your package to the buildfarm?

Timple, 2021-01-19:
Pending: https://github.com/ros/rosdistro/pull/28039

closed 2021-02-15 by Timple