nobleo/rviz_satellite #81: Add selectable default map provider

beetleskin, 2020-04-25:
To make the life of the user easier, we should add default map providers along with the tile-url field.
- check licenses/usage terms and make them transparent

**Non exhaustive list:**
- TomTom (check for free default trial key and license)
- google
- bing
- other satellite map providers?

Potentially free usage/license (to be checked)
- "OpenStreetMap": "https://a.tile.openstreetmap.org/${z}/${x}/${y}.png"
- "OpenStreetMap Germany": "https://a.tile.openstreetmap.de/${z}/${x}/${y}.png"
- "OpenStreetMap France": "http://a.tile.openstreetmap.fr/osmfr/${z}/${x}/${y}.png"
- "Humanitarian Focused OSM France": "http://a.tile.openstreetmap.fr/hot/${z}/${x}/${y}.png"
- "WikiMedia OSM": "https://maps.wikimedia.org/osm-intl/${z}/${x}/${y}.png"
- "WMFLabs Hike Bike": "https://tiles.wmflabs.org/hikebike/${z}/${x}/${y}.png"
- "WMFLabs B&W Mapnik": "https://tiles.wmflabs.org/bw-mapnik/${z}/${x}/${y}.png"
- "WMFLabs OSM No Labels": "https://tiles.wmflabs.org/osm-no-labels/${z}/${x}/${y}.png"
- "Stamen Toner": "http://a.tile.stamen.com/toner/${z}/${x}/${y}.png"
- "Stamen Water Color": "http://c.tile.stamen.com/watercolor/${z}/${x}/${y}.jpg"
- "Thunderforest Cycle": "http://tile.thunderforest.com/cycle/${z}/${x}/${y}.png"
- "Thunderforest Transport": "http://tile.thunderforest.com/transport/${z}/${x}/${y}.png"
- "Thunderforest Landscape": "http://tile.thunderforest.com/landscape/${z}/${x}/${y}.png"
- "Thunderforest Outdoors": "http://tile.thunderforest.com/outdoors/${z}/${x}/${y}.png"
- "MeMoMaps": "http://tile.memomaps.de/tilegen/${z}/${x}/${y}.png"
- "Open Topo Maps": "https://a.tile.opentopomap.org/${z}/${x}/${y}.png"
- "CartoDB Light": "https://cartodb-basemaps-a.global.ssl.fastly.net/light_all/${z}/${x}/${y}.png"
- "CartoDB Dark":  "https://cartodb-basemaps-a.global.ssl.fastly.net/dark_all/${z}/${x}/${y}.png"

schra, 2020-04-28:
I don't think that there is any tile server provider that has a license that would allow us to ship them in this plugin. AFAIK Google doesn't allow to use their URI at all and OSM restricts its usage to loading one tile per second or something like this. I also haven't checked the license of a lot of provides, so idk, maybe there is actually a provider that we could use.

beetleskin, 2020-04-29:
@dseifert any hints/ideas on that?

schra, 2020-08-02:
I just checked a few providers and all of them prohibit us to use them:

* OpenStreetMap

  From https://operations.osmfoundation.org/policies/tiles/:

  > Heavy use (e.g. distributing an app that uses tiles from openstreetmap.org) is forbidden without prior permission from the Operations Working Group. See below for alternatives.

* WikiMedia OSM

  From https://foundation.wikimedia.org/wiki/Maps_Terms_of_Use:

  > The following are prohibited on the Wikimedia Maps service:
  > 1. Excessive downloading (such as downloading significant areas of tiles for later offline usage)

* Thunderforest

  From https://www.thunderforest.com/terms/:

  > Registration is required before using any of our services.
  > Attribution must be given to both “Thunderforest” and “OpenStreetMap contributors”.

Even if there is a provider that allowed us to use them, there are more issues to tackle: When using default tile server we first need to satisfy their requirements (like showing their copyright in a corner of rviz) and second the GDPR. To satisfy the GDPR we would have to add some kind of privacy policy note at the very list (probably it's more complicated).

I think this is a lot of work for almost no benefit. Thus I'll close this.

closed 2020-08-02 by schra