nobleo/rviz_satellite #83: is it possible to convert 2D nav goal to actual gps?

KalanaRatnayake, 2020-05-15:
I have been looking at various drone trajectory planners and they expect either a GPS or xyz coordinates to functions. Since this plugin supports visualizing map on rviz, would it be possible to convert/improve "2D nav goal" functionality on rviz to  output GPS coordinates of the selected location? i can assume that the calculation depend on zoom level, (some thing like 2D nav goal -> scaled up distance -> GPS coordinate) but cannot wrap around my head a way to implement it.

dseifert, 2020-05-15:
Calculation does not depend on zoom level. I don't think it makes sense to put it into this plugin, as there would be basically zero overlapping code/functionality.

To translate from ENU (xyz coordinate, or with 2D nav goal limited to xy) to GPS (WGS84, limited to lat/lon for 2D) you could use existing ROS packages/libraries (quick googling found https://github.com/ethz-asl/geodetic_utils/blob/master/geodetic_utils/include/geodetic_utils/geodetic_conv.hpp#L166)

KalanaRatnayake, 2020-05-16:
Nice! Thanks. I'll look it up

schra, 2020-08-02:
Agreed, we will only support positions via GPS

closed 2020-08-02 by schra