nobleo/rviz_satellite #84: Wrong location in demo

Martin-Oehler, 2020-06-15:
I tried to run the demo and was shown the following location:
[image]
If I enter the same coordinates into a different site, I am shown a different location, which appears to be further in the south.

[image]
https://www.latlong.net/c/?lat=39.951121&long=-75.189930
It looks like, rviz_satellite is misreporting the position. Am I missing something here?

Best regards,
Martin Oehler

beetleskin, 2020-06-16:
Can confirm, looks like this was introduced in 5431732.

Martin-Oehler, 2020-06-16:
I can confirm, that 0028d57dceca888cad3e27239e132d15dd713cf0 does not show this behavior.

lyrem0007a, 2020-07-14:
I still have the same problem. 48.197119, 11.578062.

mishrasubhransu, 2020-07-15:
Still have the same issue [[39.326465, -76.621554]. First image is incorrect and second image is correct location. 
[image]
[image]

lyrem0007a, 2020-07-16:
I check out 0028d57 and it works well.

beetleskin, 2020-07-19:
Please check out #85, it's waiting for reviewand should fix this.

schra, 2020-08-03:
Fix is merged now

closed 2020-08-03 by schra