nobleo/rviz_satellite #89: Compilation fails on ROS Kinetic

Earthwings, 2020-10-14:
This used to work fine in ROS Kinetic, but the introduction of getTF2BufferPtr() in https://github.com/gareth-cross/rviz_satellite/commit/0e8ff386c78a1cd036b1f57022a22ceae05b4497 doesn't work there:

```
rviz_satellite/src/aerialmap_display.cpp: In member function 'virtual void rviz::AerialMapDisplay::onEnable()':
/workspace/src/external/rviz_satellite/src/aerialmap_display.cpp:98:45: error: 'class rviz::FrameManager' has no member named 'getTF2BufferPtr'
   tf_buffer_ = context_->getFrameManager()->getTF2BufferPtr();
```

I think querying it from `getTFClient()` (see http://docs.ros.org/kinetic/api/rviz/html/c++/classrviz_1_1FrameManager.html#a6eb09c3609b6591faee43483186b8730 ) would already work around it.

Would be cool if Kinetic support could stay for some more months until upstream support for it goes away in April 2021. Thanks!

schra, 2020-10-16:
Ops, that wasn't planned. Sorry. I'll add a CI to this repo soon, then this shouldn't happen anymore.

Would it be possible that you provide a PR to re-add support to kinetic?

Soverngity, 2021-01-10:
how to solve it?

Timple, 2021-01-19:
Would sticking to release 3.0.1 be an option?

That works as it seems:
```
FROM ros:kinetic
RUN apt-get update -qq && apt-get install -qqy git python-catkin-tools
RUN mkdir -p /ws/src
WORKDIR /ws
RUN bash -c 'source /opt/ros/kinetic/setup.bash && catkin init && catkin build'
RUN git clone --branch=3.0.1 https://github.com/nobleo/rviz_satellite.git src/rviz_satellite
RUN rosdep update && rosdep install --default-yes --from-paths src/
RUN bash -c 'source /ws/devel/setup.bash && catkin build'
```

If you didn't suffer from the problem solved in #88 this shouldn't be a problem.

Timple, 2022-01-06:
Kinetic EOL

closed 2022-01-06 by Timple