nobleo/rviz_satellite #90: Wrong tile positioning

blacksoul000, 2020-10-30:
Tile is always positioning with left bottom corner in (0, 0)

ROS version: 
foxy/noetic

Steps to reproduce:
Publish gps position with rqt and open plugin in rviz

RViz result:
[image]

OSM result:
[image]

p.s. got this bug on foxy but reproduced at noetic
p.p.s. if i will increase latitude for example position in rviz not changing. Only loaded tile may be changed

ceranit, 2020-11-11:
In what frame is your NavSatFix published? It should either be "map" or your should change the `AerialMapDisplay::MAP_FRAME` in line 62 of _aerialmap_display.cpp_ to whatever your frame_id is. I had a similar problem in eloquent/melodic so I hope this fix works for you too.

blacksoul000, 2020-11-12:
@ceranit 
In 'map' frame. Have you tried the same?
**RViz**
Fixed frame: map
Object URI: https://tile.openstreetmap.org/{z}/{x}/{y}.png
zoom: 17
blocks: 0
**Rqt**
frame: map
latitude: 33.893962
longitude: -77.036484

ceranit, 2020-11-12:
@blacksoul000 Yeah for me the problem was that the frames didn't match so changing it to "map" fixed it. But using those setting it looks fine to me

[image]

closed 2020-11-18 by blacksoul000