nobleo/rviz_satellite #91: Unable to get real-time map update

haritha-m, 2020-11-03:
Hello! Thanks for this great plugin.  

I'm trying to visualize the changing GPS position of a robot in real-time. I added the AerialMap plugin to rviz, and added an "axes" object to mark the base_link frame of the robot.  

Unfortunately, I am not able to visualize the position of base_link changing in real-time. When I press the "reset" button at the bottom of rviz, the position is updated; but otherwise it is sort of "stuck" in the same position.  

May I know if this behaviour is intentional, or is there a way to get the map updated in real time? 

Thank you!

beetleskin, 2020-12-01:
Its a bit hard to understand your concern. What does this plugin have to do with the Axis Display you render in RViz?

schra, 2020-12-29:
I agree, it is hard to understand what you exactly mean. Please create a short video of showcasing what you mean. Until then I will close this issue.

closed 2020-12-29 by schra