nobleo/rviz_satellite #92: Not accurate in use with robot_localization

TobiMiller, 2020-11-18:
Hey,
i am using the transform method in robot_localization to get XY from LL. I take the same origin for the transform of GeoPoints and for rviz_satellite. i set "map" as frame_id for the navsat message.

why do i get this offset? do you use different transform functions? Are the Map-Tiles generated by a different method?

[image]

beetleskin, 2020-12-01:
Hi Tobi, from the image alone its hard to judge what's going on there. 

What'd be the ground-truth here, a trajectory on the "Blückerplatz" street? Could you provide an image with ground-truth in order to judge the error a bit better? Are you on latest master version?

To narrow it down: What GPS are you using, how good is the extrinsic calibration, how good is you tf-tree? What is the NavSatFix update rate? What's your gps-map tf frame update rate?

Despite that, one important mechanism of the plugin is: The RViz tile placement only happens when a NavSat message is received. At this very moment, the messages' latitude and logitude fields are tranformed into the map-frame. If there are any offsets in your tf-tree (or offsets between tf and gps) at this very moment, you will experience offsets until the next tile is loaded (which happens when you "leave" the current tile). Maybe that helps?

schra, 2020-12-29:
Will re-open this issue when there is feedback from OP. Closed to a lack of feedback for now

closed 2020-12-29 by schra

TobiMiller, 2021-04-15:
sorry for my delayed answer. 

back to my issue. i try to explain it again:

I have a **set off GPS Coordinates** (Latitude, Longitude) (generated in JOSM - OpenStreetMap Editor),
I use Mapbox for rviz_satellite and JOSM.
I chose **one gps point as reference point**,
i am using the **husky simulation**, I've set the Initial gps coordinate of the husky gazebo gps plugin to the reference point.
i publish the **GPS coordinates converted as XY** referenced to map frame, which is the reference point (converted with https://github.com/bsb808/geonav_transform/blob/master/src/geonav_transform/geonav_conversions.py which is the same code as robot_localization is using in navsat_transform_node)

==> The GPS coordinates in RVIZ are in reference to the satellite image in a different place than in the OSM editor (see rectangles on ground)
[image]

Now I wonder if the satellite representation in your plugin is wrong or the conversion used by robot_localization.

[image]

qpc001, 2021-04-29:
> @schra
> @beetleskin
> sorry for my delayed answer.
> 
> back to my issue. i try to explain it again:
> 
> I have a **set off GPS Coordinates** (Latitude, Longitude) (generated in JOSM - OpenStreetMap Editor),
> I use Mapbox for rviz_satellite and JOSM.
> I chose **one gps point as reference point**,
> i am using the **husky simulation**, I've set the Initial gps coordinate of the husky gazebo gps plugin to the reference point (map-frame).
> i publish the **GPS coordinates converted as XY** referenced to map frame, which is the reference point (converted with https://github.com/bsb808/geonav_transform/blob/master/src/geonav_transform/geonav_conversions.py which is the same code as robot_localization is using in navsat_transform_node)
> 
> ==> The GPS coordinates in RVIZ are in reference to the satellite image in a different place than in the OSM editor (see rectangles on ground)
> [image]
> 
> [image]
> 
> Now I wonder if the satellite representation in your plugin is wrong or the conversion used by robot_localization.
> The deviation gets bigger the further you move away from the origin. Zoom Level 19 from mapbox does not work at all.
> 
> [image]

I think it's the problem with this plugin, because it's no problem when I use mapviz.