nobleo/rviz_satellite #94: Repo is unmaintained. Looking for maintainers

schra, 2020-12-30:
Due to an internal company shift, my colleagues and I will sadly no longer have the resources to maintain this repository.
 
If you want to adopt the maintenance of this repo, please contact @gareth-cross.

beetleskin, 2021-01-11:
FTR: https://discourse.ros.org/t/rviz-satellite-maintenance/18334

iamshyamganatra, 2021-01-12:
Hi, @schra, 

I'm interested in maintaining this package, I have been through the discourse post. 

Currently, I can work on 
- releasing this package for rosdistro
- adding CI/CD.

I would love to port this package to ROS2 but I don't currently have enough resources for that, but for sure I can manage in a couple of months to port this and release it for Foxy.

beetleskin, 2021-01-12:
Hey @iamshyamganatra, thanks a lot for the love and support!

It seems we've found a company that would like to take over maintenance under their roof, and Github organization. Currently we're working on transferring the repo. I'm sure they'd be super thankful for any contribution!

We'll keep you posted, and are looking forward to settle this within this week :)

Timple, 2021-01-19:
Yes, that would be us (Nobleo).

We've been using it for some of our robots for debugging. I'm also the one that requested the debian release :slightly_smiling_face: 

So first order of business would probably be releasing it. For the short-term we don't have a ROS2 migration planned. But happy to take PR's in that direction.

closed 2021-01-19 by Timple

beetleskin, 2021-01-19:
Thanks Nobleo, have fun with the tiles ;)

schra, 2021-01-19:
Also thanks again from my side!

commit 56f8a63c (2021-01-19): Revert "Add "looking for maintainers" note to README"

This reverts commit 27ab956fbcad6aa1392b1a08f2c362666909f4b4.

Closes #94