nobleo/rviz_satellite #95: confusion between TF

mpkuse, 2021-02-18:
Hello, 
I am trying to use this plugin with rviz. 

I have GPS coordinates at 100hz from sensor fusion. They are in the ENU frame (true east north and up frame). 

In this case, what is the definition of the map frame? Will setting ENU to map frame as Identity suffice? 
Any insights please?

Timple, 2021-02-18:
> They are in the ENU frame (true east north and up frame).

Frames should adhere to [REP-105](https://www.ros.org/reps/rep-0105.html)

> In this case, what is the definition of the map frame?

There should be a map frame present and the `frame_id` of your GPS message should be part of the same TF tree as the map frame.

> Is the base_link needed for this plugin?

No

mpkuse, 2021-02-19:
I publish the origin gps cords (as NavSat msg) of map frame (satellite image) only once. Is this the intended way for the plugin?

Timple, 2021-02-23:
I think the intended way is to keep streaming GPS data. As the assumption is that the robot moves.
But I guess it should work by only publishing it once as **latched** topic. Because this plugin needs to receive the data at least once.