nobleo/rviz_satellite #97: [BUG] Tiles are shifting when the robot changes the tile

TobiMiller, 2021-04-21:
**What we did:**
We selected a traffic circle as a prominent location and walked along the edge of a patch of grass with an rtk module.
We have recorded the GPS topic and we perform a Navsat to ENU conversion. We broadcast the ENU conversion from "map"->"gps_link" (gps_link is the frame_id we get from our ublox f9p driver)

**What we expect:**
That the trajectory matches the satellite image.

[image]

**What we get**

https://user-images.githubusercontent.com/32275/115557868-6b1eef80-a2b2-11eb-91f8-a99349e6dd81.mp4

TobiMiller, 2021-04-21:
I think this also addresses the following issue:
https://github.com/nobleo/rviz_satellite/issues/92

beetleskin, 2021-04-21:
Code-wise this makes sense and might be just natural for GNSS signals that have poor accuracy. But I can't rule out that this is yet a bug.

So the "map"->"gps_link" is purely produced by your GNSS receiver? I.e. the the GNSS is your only localization? Are the timestamps well synchronized? Are they running at the same frequency? What's the accuracy of your receiver, i.e. what's the jitter like?

TobiMiller, 2021-04-22:
@beetleskin 
you see the upper image? these are the RTK corrected GPS values plotted on a mapping service like ArcGIS satellite.  this is what i would expect in rviz. a little drift but it is quiet good in my opinion ;)

have you seen the whole clip from rviz? at the edges of each tile, the tiles are shifting. My TF is looking fine -> see the trajectory which gets recorded ;) the circle remains a circle and should be aligned to the traffic circle.

Timple, 2021-04-22:
The shifting at the tiles is very interesting. There doesn't seem to be much magic going on in the conversions.

Would this problem reproduce with an artifical (made up) GPS signal from a generator? And if so, only when doing circles or also straight lines?

TobiMiller, 2021-04-23:
@Timple 
Thanks for your answer.

After the weekend, I'm happy to give this a try. I just did a big lap around the block with the robot system including Robot_localization. Straight lines and curves were driven here. The shift was still present. I will prepare and upload the material on Monday.

Timple, 2021-04-23:
That would still be measured data. It might be easier to debug (isolate) the problem with a dummy publisher of the gps signal.

TobiMiller, 2021-04-27:
@Timple 
Here you go. I have done the following things:
Wrote a script that broadcasts a position as a TF. The position moves from the initial pose towards the east (x=0-1000m, y=0m). 
I then converted the position to latitude and longitude like [here] (https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/include/robot_localization/navsat_conversions.h). 
When I plot the GPS coordinates with gpsvisualizer.com, it seems to me that the values not only go towards the east, but also quite slightly towards the north. In rviz it goes perfectly to the east because x-axis=east and i broadcast y=0.

I honestly have little theoretical knowledge of geodesy to interpret all this. I tought UTM is UTM.
Here are my rosbag files.
[fake_gps_bagfiles.zip](https://github.com/nobleo/rviz_satellite/files/6385387/fake_gps_bagfiles.zip)

**(artifical)XY -> LAT/LON**
[image]

i have corrected the Navsat-Fix Message by rotating the XY Points ~-2.5 deg before converting them to LatLon.

**(artifical)XY -> (rotated) LAT/LON**
[image]

qpc001, 2021-04-28:
I meet the same bug too.

ilgazaskin, 2022-11-28:
I also have the same issue. @TobiMiller Could you explain how you rotated the XY Points 2.5deg in detail?