nobleo/rviz_satellite #98: Load images from computer

proflim123, 2021-06-01:
Hi, is it possible to load geotiffs currently stored on the computer as opposed to getting them from a URI?

Thank you. Regards, Andres

StefanKaiser-TomTom, 2021-08-30:
I recommend setting up a local WM(T)S server to provide the images. There are several open source solutions out there. Using WMS as interface is pretty common, and I think hacking some code into the plugin to work with non-standard data would not benefit the code or feature-set. E.g. http://geoserver.org/

beetleskin, 2021-08-30:
I'm afraid not.
I recommend setting up a local WM(T)S server to provide the images. There are several open source solutions out there. Using WM(T)S as interface is pretty common for raster map renderers; and I think hacking some code into the plugin to work a rather special use-case of geotiff would not benefit the code or feature-set.

If this is jut for offline-visualization, you might want to go the other way: export your robot trajectory as geojson or kml or whatever, and use e.g. plotly or mapbox; Both support direct image rendering als WMS replacement.

Checkout e.g. http://geoserver.org/ 

You might also be able to set it up without code via QGIS (https://docs.qgis.org/3.4/en/docs/training_manual/qgis_server/wms.html)

Timple, 2026-10-25:
Local WMTS server can cover this.

closed 2026-10-25 by Timple