nobleo/rviz_satellite #99: ROS2 support

Timple, 2022-01-06:
Placeholder issue.

If anyone is interested in picking this up, let me know!

vinnnyr, 2022-11-26:
There is a `ros2` branch here, but the README contains this text

> In the near future this plugin will not add support for ROS 2. For an unofficial ROS 2 fork see [blacksoul000/rviz_satellite](https://github.com/blacksoul000/rviz_satellite/tree/dashing).

I am  a little confused -- is this branch usable or not?

Timple, 2022-11-28:
We have not made any effort yet in this direction.

Perhaps the fork will open a PR here. Or somebody else.
I am not sure if the `ros2` branch is usable at the moment.

mr337, 2023-02-07:
Just curious what it would take to get ROS2 merged into this repo? A combination of https://github.com/Kettenhoax/rviz_satellite and a few tiny fixes https://github.com/Kettenhoax/rviz_satellite/pull/1 works great.

[image]

Timple, 2023-02-08:
Would you mind combining the efforts of those two in a single PR and target the ROS2 branch?

That should certainly get the ball rolling!

Timple, 2023-02-21:
Thanks all. I took the liberty of releasing the binary to the buildfarm.

Feel free to test using the testing ppa.

closed 2023-02-21 by Timple