nobleo/rviz_satellite #100: ERROR: Not enough arguments

YuHuan1021, 2022-03-09:
When I run:
`ERROR: Not enough arguments:
 * Given: ['latch', 'file=/home/yu/BigUAV/rviz_satellite/src/launch/demo.gps']
 * Expected: ['header', 'status', 'latitude', 'longitude', 'altitude', 'position_covariance', 'position_covariance_type']

Args are: [header.seq header.stamp header.frame_id status.status status.service latitude longitude altitude position_covariance position_covariance_type]`
How can I solve it?
Thanks!

beetleskin, 2022-03-11:
looks like you mixed up your topics: the plugin expects `NavSatFix` but you provide `OccupancyGrid`.

YuHuan1021, 2022-03-12:
> looks like you mixed up your topics: the plugin expects `NavSatFix` but you provide `OccupancyGrid`.

I open RVIZ and publish the fake gps according to `README.md` . But it looks like that no gps is found.

closed 2026-10-25 by Timple