nobleo/rviz_satellite #101: Failed to load plugin 'AerialMapDisplay'

EnderDragonEP, 2022-07-11:
I'm trying to intergrade satellite images to Linorobot and display them with rviz, but I keeps getting error.
`roslaunch rviz_satellite demo.launch` shows this error message:

[gps_fix-1] process has died [pid 17298, exit code 64, cmd /opt/ros/melodic/lib/rostopic/rostopic /gps/fix __name:=gps_fix __log:=/home/eilidar/.ros/log/c4f23c6a-00e4-11ed-b27c-60a4b7c94bc5/gps_fix-1.log].
log file: /home/eilidar/.ros/log/c4f23c6a-00e4-11ed-b27c-60a4b7c94bc5/gps_fix-1*.log
[ERROR] [1657525401.857090717]: PluginlibFactory: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.  Error: According to the loaded plugin descriptions the class rviz_plugins/AerialMapDisplay with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin/AerialMapDisplay rviz_plugin_tutorials/Imu

Timple, 2022-07-11:
How did you install the package?

EnderDragonEP, 2022-07-11:
I use git clone to copy this project to my workspace src folder, then run catkin_make at workspace directory.

EnderDragonEP, 2022-07-11:
I only know `catkin_make` to build, let me try using `catkin build`.
Sorry, I'm still new to ROS.

EnderDragonEP, 2022-07-11:
Using catkin-tools run `catkin build` in workspace fix this issue. Thanks for the advice!
[image]

closed 2022-07-11 by EnderDragonEP

Timple, 2022-07-11:
Great, happy to help!