nobleo/rviz_satellite #103: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.

Ed562, 2022-10-07:
Hey guys! So I'm trying to set up rviz_satellite on ubuntu 20.04 on WSL1 with ros noetic... I'm getting this error when launching the demo.launch file: 

'[ERROR] [1665163798.105047200]: PluginlibFactory: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.  Error: Could not find library corresponding to plugin rviz_plugins/AerialMapDisplay. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.'

I've checked the default library path and it's: "lib/librviz_satellite", I've checked the opt/ros/noetic/lib folder and can't find a library associated with librviz_satellite? Does anyone know how to install this? 

Rviz opens up when the demo.launch is launched with odom and map frames... the only output not showing is the AerialMapDisplay plugin which does not respond! 

Any help is of course much appreciated, Thanks!

ulterzlw, 2023-01-02:
Hi! May I ask how you set up rviz_satellite? For example, the cmd line of the setup process.

xluo6688, 2023-01-04:
Hello, ED562: I have the same issue: [ERROR] [1672799686.885784221]: PluginlibFactory. AerialMapDisplay plugin not shown. Have you got the solution? Thanks.

ulterzlw, 2023-01-04:
Hi @xluo6688 Did you source the devel/setup.bash before opening the rviz.

xluo6688, 2023-01-04:
Hello @ulterzlw , Yes, I sourced the devel/setup.bash. Thanks.

Ed562, 2023-01-04:
Hi @xluo6688 I'm sorry I ended up using mapviz instead 👍

fmiklauzic, 2023-11-16:
Hi @Ed562, I had the same issue. For me, there was poblem in defined ROS package path. It was sourced to workspace that I am working on and after catkin_make no one of the plugins have been built. I solved problem by sourcing ROS distro directly (/opt/ros/noetic/setup.bash) and deleting **build** and  **devel** folder from workspace that I am working on. Then I just ran catkin_make command. 
For me, all plugins defined in the workspace packages were built.

mulin33, 2024-11-18:
hello @Ed562 ,I had the same issue and has solved it.The **AerialMapDisplay** usually comes from a third-party ROS plugin package, such as **rviz_satellite**. First Check if the relevant plugin package is installed:`roscd rviz_satellite`  
if not installed: `sudo apt-get install ros-<distro>-rviz-satellite`  Replace <distro> with your ROS distribution (e.g., noetic or melodic). 
Then ,I can successfully run rviz without error.