nobleo/rviz_satellite #108: Fail to complie on kinetic

pdmike, 2023-02-02:
I have Ubuntu 16.04 and my ROS is kinetic. I just typed 'catkin_make' to compile rviz_satellite after copying to my workspace. I got follow errors.
[  9%] Generating src/moc_aerialmap_display.cpp
[ 18%] Generating src/moc_position_reference_property.cpp
[ 27%] Generating src/detail/moc_tile_downloader.cpp
Scanning dependencies of target rviz_satellite
[ 36%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/aerialmap_display.cpp.o
[ 54%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_position_reference_property.cpp.o
[ 54%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/tile_id.cpp.o
[ 63%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/position_reference_property.cpp.o
[ 72%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/detail/moc_tile_downloader.cpp.o
[ 81%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/ogre_tile.cpp.o
[ 90%] Building CXX object rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_aerialmap_display.cpp.o
In file included from /home/shengshi/catkin_ws/build/rviz_satellite/src/moc_position_reference_property.cpp:9:0:
/home/shengshi/catkin_ws/build/rviz_satellite/src/../../../src/rviz_satellite/src/position_reference_property.h:27:45: error: expected initializer before ‘:’ token
 class RVIZ_EXPORT PositionReferenceProperty : public TfFrameProperty
                                             ^
/home/shengshi/catkin_ws/build/rviz_satellite/src/../../../src/rviz_satellite/src/position_reference_property.h:29:20: error: expected ‘}’ before end of line
/home/shengshi/catkin_ws/build/rviz_satellite/src/../../../src/rviz_satellite/src/position_reference_property.h:29:20: error: expected declaration before end of line
In file included from /home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.cpp:15:0:
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:27:45: error: expected initializer before ‘:’ token
 class RVIZ_EXPORT PositionReferenceProperty : public TfFrameProperty
                                             ^
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:29:20: error: expected ‘}’ before end of line
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:29:20: error: expected declaration before end of line
rviz_satellite/CMakeFiles/rviz_satellite.dir/build.make:155: recipe for target 'rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_position_reference_property.cpp.o' failed
make[2]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/src/moc_position_reference_property.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
rviz_satellite/CMakeFiles/rviz_satellite.dir/build.make:103: recipe for target 'rviz_satellite/CMakeFiles/rviz_satellite.dir/src/position_reference_property.cpp.o' failed
make[2]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/src/position_reference_property.cpp.o] Error 1
In file included from /home/shengshi/catkin_ws/src/rviz_satellite/src/aerialmap_display.cpp:39:0:
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:27:45: error: expected initializer before ‘:’ token
 class RVIZ_EXPORT PositionReferenceProperty : public TfFrameProperty
                                             ^
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:29:20: error: expected ‘}’ before end of line
/home/shengshi/catkin_ws/src/rviz_satellite/src/position_reference_property.h:29:20: error: expected declaration before end of line
rviz_satellite/CMakeFiles/rviz_satellite.dir/build.make:77: recipe for target 'rviz_satellite/CMakeFiles/rviz_satellite.dir/src/aerialmap_display.cpp.o' failed
make[2]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/src/aerialmap_display.cpp.o] Error 1
CMakeFiles/Makefile2:1877: recipe for target 'rviz_satellite/CMakeFiles/rviz_satellite.dir/all' failed
make[1]: *** [rviz_satellite/CMakeFiles/rviz_satellite.dir/all] Error 2
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Anyone  knows how to solve this problem ?

Timple, 2023-02-02:
Sorry, kinetic is deprecated and this codebase is slowly moving on.

Perhaps you can try an older version like the 1.0.0 tag.

closed 2023-02-09 by pdmike

improve100, 2023-03-06:
i checkout the version ubuntu1604 kinetic 
5178c35 * master 1.0.0 1.0.0
The problem still exists.

In file included from /home/ros/rviz_satellite/src/position_reference_property.cpp:15:0:
/home/ros/rviz_satellite/src/position_reference_property.h:27:45: error: expected initializer before ‘:’ token
 class RVIZ_EXPORT PositionReferenceProperty : public TfFrameProperty
                                             ^
/home/ros/rviz_satellite/src/position_reference_property.h:29:20: error: expected ‘}’ before end of line
/home/ros/rviz_satellite/src/position_reference_property.h:29:20: error: expected declaration before end of line
In file included from /home/build/rviz_satellite/src/moc_position_reference_property.cpp:9:0:
/home/build/rviz_satellite/src/../../../src//ros/rviz_satellite/src/position_reference_property.h:27:45: error: expected initializer before ‘:’ token
 class RVIZ_EXPORT PositionReferenceProperty : public TfFrameProperty