nobleo/rviz_satellite #116: Adding SLAM maps is inconsistent with Google Maps

zyqww, 2023-08-31:
[image]

zyqww, 2023-08-31:
I put the first frame data in sensor_msgs::NavSatFix format in demo.gps, and add the point cloud receiving topic at the same time, but the displayed point cloud and the loaded Google map seem to have a rotation error? What is the reason for this? Also is only latitude and longitude considered here and no altitude? i.e. is only the xy axis considered and not the z axis?

Timple, 2023-08-31:
The pictures don't hold z-info. So that is discarded indeed.

Regarding your rotation issues: there could be a lot of reasons for that. I recommend you verify if you are adhering exactly to REP 105: https://www.ros.org/reps/rep-0105.html

zyqww, 2023-09-04:
Sorry to bother you, I checked REP 105: https://www.ros.org/reps/rep-0105.html, and still have the rotation error, which seems like an intractable problem to me. I think the "map" coordinate system of the odometry calculation method in the ros standard is "body", and the "base_link" coordinate system is "camera_init". The frame_id of the sensor_msgs/NavSatFix topic in the data set is "gps", and some related settings are shown in the following figure:
[image]
[image]
[image]
When I set the coordinate transformation to "body" and "camera_init" or "body" and "gps" in demo.launch, a bigger error will occur. The former will report an error, and the latter will move in the opposite direction.

kuanghuijun, 2024-06-26:
Could you provide an effective object URI？Thanks very much!

Timple, 2026-10-25:
No reply received

closed 2026-10-25 by Timple