nobleo/rviz_satellite #117: RVIZ crash while I change the {x}{y}{z} in tile URL

lixiangpengcs, 2023-10-16:
ItemIdentityException: Material with the name https://maps.wikimedia.org/osm-intl/19/{0}/174528.png already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[rviz2-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'

Timple, 2023-10-16:
You should not have to change the {x}{y}{z} in the url. That should go automatically when gps information is received.
What did you try to change?

WinRobotics, 2024-08-01:
I am facing similar issue for ROS2 Humble version, is this correct to view the demo?

[image]

WinRobotics, 2024-08-01:
Just need to re-enter the object URI for it to work

Timple, 2025-11-12:
Demo is fixed in #141