nobleo/rviz_satellite #118: can launch demo in ros2

Htet-Wai-Yan-HWY, 2023-11-03:
When I launch the demo.launch.xml, there is no map object url  in rviz2.After that, when I add object url manually it show `[rviz2-1] [ERROR] [1698998146.731052324] [rviz2]: ItemIdentityException: Material with the name https://tile.openstreetmap.org/%7Bz%7D/%7Bx%7D/%7By%7D.png already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[rviz2-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[rviz2-1]   what():  ItemIdentityException: Material with the name https://tile.openstreetmap.org/%7Bz%7D/%7Bx%7D/%7By%7D.png already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)` . If I try to search the url, it shows `The image "https://tile.openstreetmap.org/%7Bz%7D/%7Bx%7By%7.png" cannot be displayed because it containers errors.`

Sorry for my basic question, I am new in here.

Htet-Wai-Yan-HWY, 2023-11-11:
I have the fine tile map but I can't upload in object url. If I put url to aerial map topic , the rviz just crash. 
the error show 
`[rviz2-1] [ERROR] [1699678924.373426741] [rviz2]: ItemIdentityException: Material with the name http://localhost/tile/0/0/0.png already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
[rviz2-1] terminate called after throwing an instance of 'Ogre::ItemIdentityException'
[rviz2-1]   what():  ItemIdentityException: Material with the name http://localhost/tile/0/0/0.png already exists. in ResourceManager::add at ./.obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceManager.cpp (line 148)
`

I am using ros2 humble. Thank for your time.

Timple, 2023-12-19:
This should be fixed now on the `ros2` branch.

closed 2023-12-19 by Timple