nobleo/rviz_satellite #127: RVIZ crash

reinzor, 2024-10-03:
Backtrace:
```

#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff644526e in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#4  0x00007ffff64288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff68a5ffe in ??? () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff68bae9c in ??? () at /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff68a5a49 in std::terminate() ()
    at /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff68a5a62 in ??? () at /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007fffd6af49e7 in std::_Rb_tree<rviz_satellite::TileId, std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject>, std::_Select1st<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> >, std::less<rviz_satellite::TileId>, std::allocator<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> > >::_M_construct_node<std::piecewise_construct_t const&, std::tuple<rviz_satellite::TileId const&>, std::tuple<Ogre::SceneManager*&, Ogre::SceneNode*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, double&, double&, double&, bool&&> >
    (this=<optimized out>, __node=0x5555587cad40)
    at /usr/include/c++/13/bits/stl_tree.h:605
#10 std::_Rb_tree<rviz_satellite::TileId, std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject>, std::_Select1st<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> >, std::less<rviz_satellite::TileId>, std::allocator<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> > >::_M_create_node<std::piecewise_construct_t const&, std::tuple<rviz_satellite::TileId const&>, std::tuple<Ogre::SceneManager*&, Ogre::SceneNode*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, double&, double&, double&, bool&&> > (this=<optimized out>)
    at /usr/include/c++/13/bits/stl_tree.h:614
#11 std::_Rb_tree<rviz_satellite::TileId, std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject>, std::_Select1st<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> >, std::less<rviz_satellite::TileId>, std::allocator<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> > >::_Auto_node::_Auto_node<std::piecewise_construct_t const&, std::tuple<rviz_satellite::TileId const&>, std::tuple<Ogre::SceneManager*&, Ogre::SceneNode*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, double&, double&, double&, bool&&> >
    (__t=<optimized out>, this=<synthetic pointer>) at /usr/include/c++/13/bits/stl_tree.h:1637
#12 std::_Rb_tree<rviz_satellite::TileId, std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject>, std::_Select1st<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> >, std::less<rviz_satellite::TileId>, std::allocator<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> > >::_M_emplace_unique<std::piecewise_construct_t const&, std::tuple<rviz_satellite::TileId const&>, std::tuple<Ogre::SceneManager*&, Ogre::SceneNode*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, double&, double&, double&, bool&&> > (this=<optimized out>)
    at /usr/include/c++/13/bits/stl_tree.h:2434
#13 std::map<rviz_satellite::TileId, rviz_satellite::TileObject, std::less<rviz_satellite::TileId>, std::allocator<std::pair<rviz_satellite::TileId const, rviz_satellite::TileObject> > >::emplace<std::piecewise_construct_t const&, std::tuple<rviz_satellite::TileId const&>, std::tuple<Ogre::SceneManager*&, Ogre::SceneNode*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, double&, double&, double&, bool&&> > (this=<optimized out>) at /usr/include/c++/13/bits/stl_map.h:608
#14 rviz_satellite::AerialMapDisplay::buildTile (this=this@entry=0x555555944630, coordinate=..., offset=..., size=size@entry=23.78503069296718) at /usr/src/ros-jazzy-rviz-satellite-4.1.0-1noble.20240922.103251/src/aerialmap_display.cpp:352
#15 0x00007fffd6b1d3c2 in rviz_satellite::AerialMapDisplay::shiftMap (this=this@entry=0x555555944630, center=..., offset=..., offset@entry=..., tile_size_m=tile_size_m@entry=23.78503069296718) at /usr/src/ros-jazzy-rviz-satellite-4.1.0-1noble.20240922.103251/src/aerialmap_display.cpp:313
#16 0x00007fffd6b1d9dc in rviz_satellite::AerialMapDisplay::processMessage (this=0x555555944630, msg=Python Exception <class 'gdb.error'>: value has been optimized out
) at /usr/src/ros-jazzy-rviz-satellite-4.1.0-1noble.20240922.103251/src/aerialmap_display.cpp:260
#17 0x00007fffd6b057f6 in rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::incomingMessage (msg=Python Exception <class 'gdb.error'>: value has been optimized out
, this=0x555555944630) at /opt/ros/jazzy/include/rviz_common/rviz_common/ros_topic_display.hpp:281
#18 rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}::operator()(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>) const (__closure=<optimized out>, message=Python Exception <class 'gdb.error'>: value has been optimized out

   ) at /opt/ros/jazzy/include/rviz_common/rviz_common/ros_topic_display.hpp:223
#19 std::__invoke_impl<void, rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}&, std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const> >(std::__invoke_other, rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}&, std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>&&) (__f=<optimized out>)
    at /usr/include/c++/13/bits/invoke.h:61
#20 std::__invoke_r<void, rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}&, std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const> >(rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}&, std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>&&) (__fn=<optimized out>) at /usr/include/c++/13/bits/invoke.h:111
#21 std::_Function_handler<void (std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>), rviz_common::RosTopicDisplay<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >::subscribe()::{lambda(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>&&) (__functor=<optimized out>, __args#0=<optimized out>) at /usr/include/c++/13/bits/std_function.h:290
#22 0x00007fffd6b0c2b2 in std::function<void (std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)>::operator()(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>) const
    (__args#0=std::shared_ptr<const sensor_msgs::msg::NavSatFix_<std::allocator<void> >> (empty) = {...}, this=<optimized out>) at /usr/include/c++/13/bits/std_function.h:591

#23 rclcpp::AnySubscriptionCallback<sensor_msgs::msg::NavSatFix_<std::allocator<void> >, std::allocator<void> >::dispatch<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >(std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}::operator()<std::function<void (std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)>&>(std::function<void (std::shared_ptr<sensor_msgs::msg::NavSatFix_<std::allocator<void> > const>)>&) const (__closure=<optimized out>, callback=<optimized out>)
    at /opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:533
[... cut ...]
#45 0x00007ffff7335279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#46 0x00007ffff72d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#47 0x00007ffff72df3e8 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#48 0x0000555555556c97 in main (argc=6, argv=0x7fffffff9b28) at /usr/src/ros-jazzy-rviz2-14.1.5-1noble.20240922.104940/src/main.cpp:91
```

Timple, 2025-11-12:
Since it's pointer related, might be fixed by: #132

Timple, 2026-10-25:
Probably fixed

closed 2026-10-25 by Timple