nobleo/rviz_satellite #130: OpenStreetMap tile rotation error around z

ProKaroly, 2025-02-19:
At my location (Lat: 47.47800, Lon: 19.05666), I observed that the OpenStreetMap tile appears to be misaligned, exhibiting a rotation of approximately 3° around the Z-axis. To address this, I modified the aerialMap_display.cpp file by adding the following code:

`  Ogre::Quaternion rotation_offset(Ogre::Degree(3), Ogre::Vector3::UNIT_Z);

// Combine the offset with the existing orientation (order matters)
  Ogre::Quaternion new_orientation = rotation_offset * orientation_to_map;

// Apply the negative of the new orientation to the scene node
  scene_node_->setOrientation(-new_orientation);`

This adjustment effectively corrects the rotation issue at my location. However, I have two questions:

    Is the 3° rotation error a constant offset for all locations, or is it specific to my geographic location?
    Is this the best method for applying a rotation correction to the tile image, or is there a more robust solution available?

Timple, 2025-02-19:
Ow dear, that's bad news.

I'm quite sure we can't just rotate all maps with 3 degrees...

What is your other source? And how are you sure this other source is correct?

ProKaroly, 2025-02-19:
I have a series of UTM measurements that are continuously published on a ROS2 Odometry topic. The map tiles being loaded are fixed to an origin at Lat: 47.47800, Lon: 19.05666. I observed that as the UTM measurements get closer to the tile's origin (or center), the apparent error diminishes. However, as the measurements move further from the origin, the error increases. This behavior led me to suspect that there is a rotational misalignment between the map and the measurements.

It is possible that my calculation of UTM coordinates from latitude and longitude might be incorrect. Although I’m not entirely sure where the error lies, I found that applying a 3° rotation to the map effectively corrects the issue. This doesn't necessarily mean the map is wrong; rather, it's an interim solution to compensate for what appears to be a rotation mismatch.

Timple, 2026-10-25:
Assuming fixed by #139

closed 2026-10-25 by Timple